ThomasGerstenberg/blatann #22: Add support for pc-ble-driver-py v0.12.0

ThomasGerstenberg, 2019-07-16:
pc-ble-driver-py v0.12 has been updated to use a newer version of softdevice (v5) and connectivity firmware (SDK v14.x).
- Will need to rework the entire `nrf_driver` layer as the API calls into the softdevice changed nontrivially
- New version removes support for py2 (NordicSemiconductor/pc-ble-driver-py#113), so this lib will also drop py2 support and transition to py3 (3.7+ unless convinced otherwise)

Will need to major rev the lib as it will not be backwards compatible

kafouros, 2019-10-23:
Hi Thomas! It is great to know that you are working on that update. Do you have an ETA for it? Many thanks.

ThomasGerstenberg, 2019-10-23:
@kafouros Unfortunately I haven't had time to work on it yet, probably won't be ready for a couple months.

edglue, 2019-11-14:
Hey Thomas. I work with @kafouros and have found your library very useful for quickly implementing GATTS databases in simulators, thank you for your efforts. If there's anything I can do to help add features, please let me know. 

We've recently seen problems loading the underlying pc-ble-driver through pytest (with or without your abstraction) in later versions of Python v2.7. Have you seen any such issues, and if so, have you found any workarounds?

ThomasGerstenberg, 2019-11-15:
Hi @edglue, can't say I've seen that issue myself, though I haven't tried running with pytest. I'm running Python 2.7.16 with no issues and have used several other patch versions before it. I see that 2.7.17 was released about a month ago, is that the version you're testing with?

I also remember there was a Python version that had a regression when trying to load a DLL with unicode characters or something (v2.7.12 or 2.7.13, somewhere around there) that caused some issues. I'd suggest testing with 2.7.16 and see if that works, otherwise it may be an environment issue

ThomasGerstenberg, 2019-11-21:
Update: porting to new version is going well with the scanner, broadcaster, and all peripheral examples up and running. Next up is working through the central side of things. Can follow progress on the `py3-softdevicev5` branch.

If you want to test out the changes you can pip install using 

`pip3 install git+https://github.com/ThomasGerstenberg/blatann@py3-softdevicev5`

The largest breaking change is in Py3 `str` being unicode and now need to use `bytes`. For backwards compatibility you can still update characteristic values using strings; characteristics have a new `string_encoding` property used to convert to bytes. This defaults to "utf8"

edglue, 2019-11-21:
Yes, I was testing with v2.7.17. Great, I'll try creating an environment with v2.7.16 and see if that helps. Your information about loading the DLL doesn't seem a million miles away from the problems we're seeing, so perhaps it's reared its head.

Glad to hear the porting is going well. I will let you know when I've had a chance to test your development version.

ThomasGerstenberg, 2020-02-26:
0.3.0 release candidate published and available through pip. Testing looks good but haven't gotten too many cycles through it, will remove RC status once I have time to test a bit more

closed 2020-02-26 by ThomasGerstenberg