ThomasGerstenberg/blatann #40: Service discovery fails

DjordjeR, 2020-06-08:
Hi, I am using your library and I've come across of something I think might be bug. Service discovery fails when I try it with my BLE device. What I managed to conclude fromt he code is that instead of sending the end_handle when there are no more services, you get BLEGattStatusCode.attribute_not_found.

What I did to remedy this problem is update the service_dicovery.py, and it seems to work at least in my case. You can checkout the link in [1] (line: 286) on how the pc-ble-driver-py guys did this.

```python
...
        if event.status == nrf_events.BLEGattStatusCode.attribute_not_found:
            end_handle = 0xFFFF
        else:
            self._state.services.extend(event.services)
            end_handle = event.services[-1].end_handle

        if end_handle != 0xFFFF:
...
```

After this the service discovery fails to find UUIDs for characteristics, I had to add register_uuid to from_nrf_char ... but I get all 0000.

This is how UUIDs look like with pc-ble-driver py:
[image]

And this is how I get them:

[image]

And this is NRF Connect Application for Android:
[image]

The problem it would seem is that the fact that this device uses UUIDs with different base for service and custom char for that service.

Hope some of this helps!

[1] https://github.com/NordicSemiconductor/pc-ble-driver-py/blob/b8d22a6e4c146fedc13c8d5a330febe8a0df5795/pc_ble_driver_py/ble_adapter.py#L286

ThomasGerstenberg, 2020-06-08:
Thanks @DjordjeR, both of these issues look like a miss on my end. 

Characteristic UUID discovery should be happening so I'll look further into this, and if you are able to share blatann logs of the service discovery portion that would be greatly helpful as well

DjordjeR, 2020-06-09:
Yes, you can find the log file in the [1], I also did some searching, and in the log files you will find

`ERROR:blatann.gatt.service_discovery:Characteristic UUID not 16 bytes: []`, this happens because in service_discovery.py  `uuid_bytes = event.data[3:]` is 0, what I could not figure out is why that was happening. However I did need to add the workaround I mentioned in previous comment to even get here.

[1][lb_logs.txt](https://github.com/ThomasGerstenberg/blatann/files/4750075/lb_logs.txt)

ThomasGerstenberg, 2020-06-09:
I did a quick test with random UUIDs for service and characteristics and it works as expected: 

```
[2020-06-09 00:44:43,800] [MainThread] [blatann.gatt.service_discovery.start:405] [INFO]: Starting discovery..
[2020-06-09 00:44:43,801] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [COM13] ble_gattc_prim_srvc_disc(0, None, 1)
[2020-06-09 00:44:43,815] [COM13_Event] [blatann.device.on_driver_event:30] [DEBUG]: Got NRF Driver event: GattcEvtPrimaryServiceDiscoveryResponse(conn_handle=0, status=<BLEGattStatusCode.success: 0>, services=[BLEGattService(uuid=0x1800, start_handle=1, end_handle=9), BLEGattService(uuid=0x1801, start_handle=10, end_handle=10)])
[2020-06-09 00:44:43,816] [COM13_Event] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [COM13] ble_gattc_prim_srvc_disc(0, None, 11)
[2020-06-09 00:44:43,830] [COM13_Event] [blatann.device.on_driver_event:30] [DEBUG]: Got NRF Driver event: GattcEvtPrimaryServiceDiscoveryResponse(conn_handle=0, status=<BLEGattStatusCode.success: 0>, services=[BLEGattService(uuid=0x00000000000000000000000000000000, start_handle=11, end_handle=65535)])
[2020-06-09 00:44:43,831] [COM13_Event] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [COM13] ble_gattc_read(0, 11)
[2020-06-09 00:44:43,845] [COM13_Event] [blatann.device.on_driver_event:30] [DEBUG]: Got NRF Driver event: GattcEvtReadResponse(conn_handle=0, status=<BLEGattStatusCode.success: 0>, error_handle=0, attr_handle=11, offset=0, data=')¢Ø\x80\x94j\x17ë\x94\x99À5iu\x956')
[2020-06-09 00:44:43,846] [COM13_Event] [blatann.gatt.service_discovery._on_service_uuid_read:152] [INFO]: Got gattc read: GattcEvtReadResponse(conn_handle=0, status=<BLEGattStatusCode.success: 0>, error_handle=0, attr_handle=11, offset=0, data=')¢Ø\x80\x94j\x17ë\x94\x99À5iu\x956')
[2020-06-09 00:44:43,846] [COM13_Event] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [COM13] ble_vs_uuid_add(<blatann.nrf.nrf_types.generic.BLEUUIDBase object at 0x000002DC784DA0C8>,)
[2020-06-09 00:44:43,849] [COM13_Event] [blatann.gatt.service_discovery._on_service_uuid_read:164] [INFO]: Discovered UUID: 0x3695756935C09994EB176A9480D8A229
[2020-06-09 00:44:43,849] [COM13_Event] [blatann.peer.driver_event_unsubscribe:418] [DEBUG]: Unsubscribing <bound method _ServiceDiscoverer._on_primary_service_discovery of <blatann.gatt.service_discovery._ServiceDiscoverer object at 0x000002DC784BE688>> (<function Peer._check_driver_event_connection_handle_wrapper.<locals>.wrapper at 0x000002DC78491DC8>)
[2020-06-09 00:44:43,850] [COM13_Event] [blatann.peer.driver_event_unsubscribe:418] [DEBUG]: Unsubscribing <bound method _ServiceDiscoverer._on_service_uuid_read of <blatann.gatt.service_discovery._ServiceDiscoverer object at 0x000002DC784BE688>> (<function Peer._check_driver_event_connection_handle_wrapper.<locals>.wrapper at 0x000002DC784954C8>)
[2020-06-09 00:44:43,850] [COM13_Event] [blatann.gatt.service_discovery._on_service_discovery_complete:372] [INFO]: Service Discovery complete
[2020-06-09 00:44:43,850] [COM13_Event] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [COM13] ble_gattc_char_disc(0, 1, 9)
[2020-06-09 00:44:43,867] [COM13_Event] [blatann.device.on_driver_event:30] [DEBUG]: Got NRF Driver event: GattcEvtCharacteristicDiscoveryResponse(conn_handle=0, status=<BLEGattStatusCode.success: 0>, characteristics=[BLEGattCharacteristic(uuid=0x2A00, handle_decl=2, handle_value=3, data_decl=None, data_value=None, end_handle=None), BLEGattCharacteristic(uuid=0x2A01, handle_decl=4, handle_value=5, data_decl=None, data_value=None, end_handle=None), BLEGattCharacteristic(uuid=0x2A04, handle_decl=6, handle_value=7, data_decl=None, data_value=None, end_handle=None), BLEGattCharacteristic(uuid=0x2AA6, handle_decl=8, handle_value=9, data_decl=None, data_value=None, end_handle=None)])
[2020-06-09 00:44:43,868] [COM13_Event] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [COM13] ble_gattc_char_disc(0, 10, 10)
[2020-06-09 00:44:43,883] [COM13_Event] [blatann.device.on_driver_event:30] [DEBUG]: Got NRF Driver event: GattcEvtCharacteristicDiscoveryResponse(conn_handle=0, status=<BLEGattStatusCode.attribute_not_found: 266>, characteristics=[])
[2020-06-09 00:44:43,883] [COM13_Event] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [COM13] ble_gattc_char_disc(0, 11, 65535)
[2020-06-09 00:44:43,898] [COM13_Event] [blatann.device.on_driver_event:30] [DEBUG]: Got NRF Driver event: GattcEvtCharacteristicDiscoveryResponse(conn_handle=0, status=<BLEGattStatusCode.success: 0>, characteristics=[BLEGattCharacteristic(uuid=0x00000000000000000000000000000000, handle_decl=12, handle_value=13, data_decl=None, data_value=None, end_handle=None), BLEGattCharacteristic(uuid=0x00000000000000000000000000000000, handle_decl=14, handle_value=15, data_decl=None, data_value=None, end_handle=None)])
[2020-06-09 00:44:43,898] [COM13_Event] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [COM13] ble_gattc_char_disc(0, 15, 65535)
[2020-06-09 00:44:43,912] [COM13_Event] [blatann.device.on_driver_event:30] [DEBUG]: Got NRF Driver event: GattcEvtCharacteristicDiscoveryResponse(conn_handle=0, status=<BLEGattStatusCode.attribute_not_found: 266>, characteristics=[])
[2020-06-09 00:44:43,912] [COM13_Event] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [COM13] ble_gattc_read(0, 12)
[2020-06-09 00:44:43,928] [COM13_Event] [blatann.device.on_driver_event:30] [DEBUG]: Got NRF Driver event: GattcEvtReadResponse(conn_handle=0, status=<BLEGattStatusCode.success: 0>, error_handle=0, attr_handle=12, offset=0, data='\x08\r\x00í{-¾\x1d [... cut ...]
GattcEvtReadResponse(conn_handle=0, status=<BLEGattStatusCode.success: 0>, error_handle=0, attr_handle=24, offset=0, data='\x06')
```

If you have access to the server's codebase I'd suggest getting some logs from that end. Otherwise, the workaround (which you've already mentioned) would be to register the UUIDs prior to service discovery through the device's [uuid_manager](https://github.com/ThomasGerstenberg/blatann/blob/f0b319fa009b8f527c96d7e33bbc299494f541c0/blatann/device.py#L104) attribute

DjordjeR, 2020-06-09:
Unfortunately I do not have access to the source code for the server device. It is a BLE light bulb I ordered for testing. I did notice that the data recieved from server does not look correct, however the NRF Connect on android, and blub app on android are working correctly, also when I use service_dicovery from pc-ble-driver-py I works partially correct. 

Also notice this part, that I also needed to avoid with **_BLEGattStatusCode.attribute_not_found_** is is possible that by doing this I broke something with characteristics discovery, maybe the wrong handles are being read. Becuse without doing that the service discovery would fail right away, as the services are not ending in 0xFFFF.

Not sure what else it might be. 

```
DEBUG:blatann.device:Got NRF Driver event: GattcEvtPrimaryServiceDiscoveryResponse(conn_handle=0, status=<BLEGattStatusCode.success: 0>, services=[BLEGattService(uuid=0x00000000000000000000000000000000, start_handle=23, end_handle=26)])
DEBUG:blatann.nrf.nrf_driver:[COM9] ble_gattc_prim_srvc_disc(0, None, 27)
DEBUG:blatann.device:Got NRF Driver event: GattcEvtPrimaryServiceDiscoveryResponse(conn_handle=0, status=<BLEGattStatusCode.attribute_not_found: 266>, services=[])
DEBUG:blatann.nrf.nrf_driver:[COM9] ble_gattc_read(0, 15)
```

ThomasGerstenberg, 2020-06-09:
I'm currently working on a fix for the `attribute_not_found` issue and should have that in shortly. I don't expect that issue to affect the UUID discovery in any way.

The reason this isn't working properly in blatann and pc-ble-driver-py is a limitation of the Nordic API for service discovery and how they handle UUIDs. In order to save memory, instead of using 128-bit data everywhere they condense UUIDs down into the 16-bit representation plus a 16-bit index which is a lookup into a table of vendor-specific UUID bases (built by vs_uuid_register). This effectively normalizes all UUIDs to use 32 bits instead of varying between 16 and 128 bit.

Characteristic discovery uses the `Read By Type` ATT request which returns the permissions, UUID, etc. for the characteristic. What happens is Nordic sees the 128-bit UUID, doesn't recognize the base in its table of registered custom UUIDs, so returns an `UNKNOWN` index. The actual UUID value is not provided to the application the callback. 

This isn't a limitation on Android/iOS, hence why they don't see this issue. And I believe why pc-ble-driver-py looks differently is due to their re-discovery of the service after registering the UUID. Once the service is rediscovered with the custom UUID registered, when discovering characteristics for that service it assumes all characteristics have the same base as the service

During characteristic discovery in blatann, it takes all the characteristic UUIDs reported as unknown by the BLE stack and issues ATT Read Requests to read the UUID value, then registers the UUID manually. It seems in your case the BLE stack in the peripheral doesn't seem to correctly handle Read Requests for the Characteristic Declaration attribute. 

Unfortunately the Nordic API also doesn't provide a function to issue a `Read By Type` request in its raw form, otherwise I'd switch the characteristic UUID discovery to use that. Without access to the peripheral's codebase it seems you are stuck with the workaround of registering the characteristic UUIDs prior to service discovery using the UUID Manager.

DjordjeR, 2020-06-09:
Hi, thank you very much! 

Your explanation is in line with I have seen and it does make sense. My only concern was that by doing the workaround for `attribute_not_found ` I might have broken something, for example, read wrong handles, and that is the reason that I was getting the wrong values when discovering characteristic, however after thorough inspection that does not seem to be the case. After you have pushed an update I will test it again with the same device and let you know. 

You have been most helpful, thanks!

ThomasGerstenberg, 2020-06-10:
@DjordjeR the fix for `attribute_not_found` is added into branch `issue-40-service-discovery-failure`.
 Unfortunately the other part of the issue (UUID discovery) cannot be fixed and you will need to register those characteristics with the UUID Manager prior to service discovery

DjordjeR, 2020-06-10:
As expected your fix does work, but the other problem persits. Thanks!