ThomasGerstenberg/blatann #70: Characteristic indication timeout due to _NotificationManager bug

cesarvandevelde, 2020-12-23:
There appears to be an issue with characteristic indications in the current version of Blatann.
Waiting on a characteristic `notify()` raises a timeout error when `CCCD` is set to `0x02`, even though `GattsEvtHandleValueConfirm` is received from the nRF device:

```Python
payload = bytes.fromhex("aa bb cc dd 11 22 33 44")
waitable = my_characteristic.notify(payload)
waitable.wait(0.2)
```

Results in:
```
debug      blatann.nrf.nrf_driver - [/dev/cu.usbmodemD4F615BEE53A2] ble_gatts_hvx(1, BLEGattsHvx(handle=13, type=<BLEGattHVXType.indication: 2>, offset=0, data=b'\xaa\xbb\xcc\xdd\x11"3D'))
debug              blatann.device - Got NRF Driver event: GattsEvtHandleValueConfirm(conn_handle=1, attr_handle=13)
debug              blatann.device - Got NRF Driver event: GattsEvtWrite(conn_handle=1, attr_handle=16, uuid=0x0000XXXX000000000000000000000000, write_op=<BLEGattsWriteOperation.write_cmd: 2>, auth_required=True, offset=0, data=b'xxxxxxxxxxxxxxxxxxxxxxxxx')
debug    ...test.xxxxxxxxxxxxxxxx - From device: xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx xx
info     ...test.xxxxxxxxxxxxxxxx - Done
debug      blatann.nrf.nrf_driver - [/dev/cu.usbmodemD4F615BEE53A2] ble_gap_disconnect(1, <BLEHci.remote_user_terminated_connection: 19>)
debug              blatann.device - Got NRF Driver event: GapEvtDisconnected(conn_handle=1, reason=<BLEHci.local_host_terminated_connection: 22>)
info      blatann.gap.advertising - Starting advertising, params: 'BLEGapAdvParams'(type: <BLEGapAdvType.connectable_undirected: 0>, interval: 100ms, timeout: 0s), auto-restart: True
debug      blatann.nrf.nrf_driver - [/dev/cu.usbmodemD4F615BEE53A2] ble_gap_adv_start('BLEGapAdvParams'(type: <BLEGapAdvType.connectable_undirected: 0>, interval: 100ms, timeout: 0s), 1)
info     ...test.xxxxxxxxxxxxxxxx - Disconnected from device
debug      blatann.nrf.nrf_driver - [/dev/cu.usbmodemD4F615BEE53A2] close()
Traceback (most recent call last):
  File "/Users/xxxxx/xxxxxxxxx/xxxxxxxxxxx/venv/lib/python3.8/site-packages/blatann/waitables/waitable.py", line 33, in wait
    results = self._queue.get(timeout=timeout)
  File "/usr/local/Cellar/python@3.8/3.8.6_2/Frameworks/Python.framework/Versions/3.8/lib/python3.8/queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/Cellar/python@3.8/3.8.6_2/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/local/Cellar/python@3.8/3.8.6_2/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Users/xxxxx/xxxxxxxxx/xxxxxxxxxxx/hil_test/__main__.py", line 5, in <module>
    main(verbose=True)
  File "/Users/xxxxx/xxxxxxxxx/xxxxxxxxxxx/hil_test/__init__.py", line 36, in main
    waitable.wait(0.2)
  File "/Users/xxxxx/xxxxxxxxx/xxxxxxxxxxx/venv/lib/python3.8/site-packages/blatann/waitables/event_waitable.py", line 23, in wait
    res = super(EventWaitable, self).wait(timeout, exception_on_timeout)
  File "/Users/xxxxx/xxxxxxxxx/xxxxxxxxxxx/venv/lib/python3.8/site-packages/blatann/waitables/waitable.py", line 40, in wait
    raise TimeoutError("Timed out waiting for event to occur. "
blatann.exceptions.TimeoutError: Timed out waiting for event to occur. Waitable type: IdBasedEventWaitable
```

From what I can tell, the issue stems from a bug in _NotificationManager, and can be fixed with the following change in `gatt/managers.py`:

```diff
    def _on_hvc(self, driver, event):
        notification = self._pop_task_in_process()
        if notification:
+           notification.notify_complete(NotificationCompleteEventArgs.Reason.SUCCESS)
            self._task_completed(notification)
```

Apologies for sparse code example and redacted logs. I cannot share more information due to the nature of the project.

ThomasGerstenberg, 2020-12-23:
Thanks @cesarvandevelde, I'll take a look into this

ThomasGerstenberg, 2020-12-24:
Thank you @cesarvandevelde! Looks like I introduced that bug when I refactored those managers. Added tests and verified your fix works

closed 2020-12-24 by ThomasGerstenberg

cesarvandevelde, 2020-12-25:
Cheers, thanks for the quick fix!

PR #71 (2020-12-24): Fixes issue with indication waitables not notifying correctly
Fixes #70