ThomasGerstenberg/blatann #75: Repeated connection to NRF52 fails with NrfError.rpc_h5_transport_state

juhhov, 2021-02-25:
Script to reproduce the issue:
```
from blatann import BleDevice

dev = '/dev/ttyACM0'

def create():
    print('create')
    ble = BleDevice(dev)
    ble.open()
    return ble

def destroy(ble):
    print('destroy')
    ble.close()

ble = create()
destroy(ble)
ble = create()
destroy(ble)
```

Second open() raises `pc_ble_driver_py.exceptions.NordicSemiException: Failed to open. Error code: NrfError.rpc_h5_transport_state`.

Package versions:
```
blatann==0.3.6
pc-ble-driver-py==0.15.0
```

NRF52 dongle firmare:
`connectivity_4.1.2_usb_with_s132_5.1.0.hex` from `pc-ble-driver-py` package.

ThomasGerstenberg, 2021-02-25:
Thanks for reporting, this is an issue I've known about for awhile but never formally opened a ticket for it. 

The issue is that when the device is closed, the micro is reset to ensure no more bluetooth activity after that point. With the USB-based firmware images, the USB device goes away completely and has to be re-enumerated before opening again, so a delay is needed between close and open. Using serial-based firmware images with the nordic dev kit don't seem to have this issue.

The delay amount I found depends on the OS and other factors, so baking in a delay on either open or close is non-ideal and some form of retry mechanism is likely required.

A simple workaround for now is to sleep between open and close which is what I do for my integrated tests [here](https://github.com/ThomasGerstenberg/blatann/blob/45307beea5d9c30e9634500889892f56b1520f13/tests/integrated/base.py#L78)

teeheee, 2021-11-04:
Hi,

I implemented a possible fix for the issue. Its a retry loop with a scanner for the nrf52 dongle based on libusb to not use a long delay. If its ok I would create a pullrequest. Please let me know if you have any further suggestions.

regards
Alex

mdxs, 2023-11-08:
Check my comment with #103 providing a script that may help improve the situation with the nRF52840-Dongle (PCA10059).