ThomasGerstenberg/blatann #77: Central: Multiple-Connections

trembel, 2021-03-11:
Is it possible to create a multiple-connection central using blatann? If yes, could you create an example or lead me to the documentation (as I couldn't find it). If no, what is needed in order to get it working?

Best,
Trembel

ThomasGerstenberg, 2021-03-12:
Yes it does support multiple connections to peripheral devices. I don't have any examples currently and documentation doesn't spell it out very well (overall the documentation needs a lot of work), but in order to enable it set `ble_device.configure(max_connected_peripherals=x, max_secured_peripherals=x)` alongside any other config params after opening the device. If you're not pairing/bonding, can leave out `max_secured_peripherals` parameter. [link](https://github.com/ThomasGerstenberg/blatann/blob/86366bc38d84f3ddb22314aa1ca50ee28be8b916/blatann/device.py#L145). The practical limit here is 8, but theoretical limit IIRC is 20 (requires finely-tuned connection and scan parameters).

After that the usage is the same as the central example, when you scan and connect to a peripheral you'll get a `Peer` object for that connection. From there you can discover the peer's services and perform characteristic reads/writes.

If I have time I can create a basic example but probably won't be able to get to it for awhile

ThomasGerstenberg, 2023-11-19:
Example to be created as a part of #174

closed 2023-11-19 by ThomasGerstenberg