ThomasGerstenberg/blatann #92: Use of more than 4 peripherals

andreh1111, 2021-12-09:
Hi, I'm trying to link with more than 4 peripherals, but I get the error:
pc_ble_driver_py.exceptions.NordicSemiException: Failed to ble_enable. Error code: NrfError.no_mem

Do you have a solution for this problem?
Thank you very much for your work.

ThomasGerstenberg, 2021-12-21:
Can you confirm that you're configuring the device with the appropriate number of peripherals (and secured peripherals)? https://github.com/ThomasGerstenberg/blatann/blob/master/blatann/device.py#L145-L147

Additionally if you could turn on blatann logs that'll help narrow down what the issue is. Easiest way is to call `import blatann.utils; blatann.utils.setup_logger("blatann")`

I personally have not tested with more than 2 peripherals. If I get a chance I'll see if I can grab a bunch of cheap OTS BLE devices and see how many I can connect to at once

andreh1111, 2021-12-23:
Hi Thomas, thank you for your reply. 
Yes, I confirm you I'm configuring properly the device. This is the code lines of interest:
```
ble_device = BleDevice(serial_port)
ble_device.configure(max_connected_peripherals=5, max_secured_peripherals=5)
ble_device.open()
```
I set the logger as you asked, this is the output:
```
C:\Users\andre\Desktop>python central.py
[2021-12-23 16:49:59,703] [MainThread] [blatann.gap.default_bond_db.load:33] [INFO]: Loaded Bond DB 'C:\Users\andre\.blatann\bonding_db.pkl'
[2021-12-23 16:49:59,703] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [COM12] open()
[2021-12-23 16:50:01,005] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [COM12] ble_conn_configure(<blatann.nrf.nrf_types.config.BleConnConfig object at 0x0000017FFB57E7F0>,)
[2021-12-23 16:50:01,008] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [COM12] ble_enable(<blatann.nrf.nrf_types.config.BleEnableConfig object at 0x0000017FFB593760>,)
Traceback (most recent call last):
  File "central.py", line 88, in <module>
    main(element.device)
  File "central.py", line 30, in main
    ble_device.open()
  File "C:\Users\andre\AppData\Local\Programs\Python\Python38\lib\site-packages\blatann\device.py", line 188, in open
    self.ble_driver.ble_enable(self._ble_configuration)
  File "C:\Users\andre\AppData\Local\Programs\Python\Python38\lib\site-packages\blatann\nrf\nrf_driver.py", line 80, in wrapper
    raise NordicSemiException('Failed to {}. {}'.format(wrapped.__name__, err_string), err_code)
pc_ble_driver_py.exceptions.NordicSemiException: Failed to ble_enable. Error code: NrfError.no_mem
```
This error occurs only if I set max_connected_peripherals > 4. Note that the error was obtained during the open() method, so there may be no need to test N peripherals to face this issue since no connection has been established with any device yet.
I will stay available for any further information.
Thank you for your amazing work.

Andrea

ThomasGerstenberg, 2022-01-21:
Thanks, I'll take a look. It could be a limit of the firmware image if it's reporting `no_mem`, though I'd expect that it should be able to support 4+ simultaneous devices at once

mdxs, 2023-09-26:
Out of curiosity, I just tried the example and can confirm the same result:
`pc_ble_driver_py.exceptions.NordicSemiException: Failed to ble_enable. Error code: NrfError.no_mem`

occurs when using:
`ble_device.configure(max_connected_peripherals=5, max_secured_peripherals=5)`

while this doesn't occur when using:
`ble_device.configure(max_connected_peripherals=4, max_secured_peripherals=4)`

It appears that in either case there is serial communication going on from `pc_ble_driver_py` to the nRF52840 dongle when I set `log_driver_comms=True` on the BleDevice. With the 5/5 it just ends in the exception.

andreh1111, 2023-09-26:
Hi @mdxs
I don't think this is a firmware limitation, since I can use the same dongle with the same firmware on _nRFConnect for Desktop_ to connect more than 4 devices.

mdxs, 2023-09-26:
Interesting, as then it is more likely that it is a limitation in the `pc-ble-driver-py` (or the underlying `pc-ble-driver`). As I noted before, there appears to be serial communication going on and then something crashes and raising the exception.

ThomasGerstenberg, 2023-09-26:
the `no_mem` return code is the firmware indicating that for the provided configuration, it doesn't have enough RAM to accommodate everything requested. The firmware statically allocates all the necessary blocks when the device is configured, and if it cant fit everything it'll error out.

I can play around with settings in the next week or so, but you could try playing around with these settings to see if you can get more than 4 peripherals:

- Hardware tx/rx queue sizes (BleDevice constructor params, can reduce to 1)
- Attribute table size (default is 1408, if you are only using this as a central you could try reducing to the minimum 248)
- Max MTU size
- Vendor specific UUID count

mdxs, 2023-09-26:
There indeed seems to be a correlation to the "Hardware tx/rx queue sizes", as the following worked for me (just setting it up, no usage whatsoever):

```
ble_device = BleDevice(serial_port, log_driver_comms=True,
					   notification_hw_queue_size=8,
					   write_command_hw_queue_size=8)
ble_device.event_logger.suppress(nrf_events.GapEvtAdvReport)
ble_device.configure(max_connected_peripherals=8)
ble_device.open()
```

Increasing the max_connected_peripherals to 9 in this script would raise the `no_mem` again.

mdxs, 2023-09-26:
The following shows `max_connected_peripherals = 12`, reducing the "Attribute table size" (less important?) and the "Max MTU size" (does seem important):

```
ble_device = BleDevice(serial_port, log_driver_comms=True,
					   notification_hw_queue_size=8,
					   write_command_hw_queue_size=8)
ble_device.event_logger.suppress(nrf_events.GapEvtAdvReport)
ble_device.configure(max_connected_peripherals=12,
					 attribute_table_size=248,
					 att_mtu_max_size=150)
ble_device.open()
```

Please note that I've no idea about the impact on using these settings, just testing if it can be setup as such with the hints provided by @ThomasGerstenberg to "play around" with some settings.

Note that I haven't tried to do something with the "Vendor specific UUID count", but I'm calling it a day.

ThomasGerstenberg, 2023-09-27:
The largest contributors are going to be the queue sizes and MTU size since they're multiplied by the # concurrent connections. Rough calculation of variable RAM usage:

```
attribute_table_size + 
(vendor_specific_uuid_count * 20ish) +
(max_connected_centrals * notification_hw_queue_size * att_mtu_max_size) +
(max_connected_peripherals * write_command_hw_queue_size * att_mtu_max_size) + 
(max_secured_peripherals * some relatively small constant)
```

You could likely set the queue size to 2 and still have performant software, IIRC in my testing I never saturated the queue when looking at max data throughput

ThomasGerstenberg, 2023-09-30:
I've added this information to the troubleshooting section of the docs including some hints for config settings to play around with and a "magic" value to aim for in terms of ram usage

PR #149 (2023-09-30): Adds section to troubleshooting for no_mem errors when increasing # connections
Fixes #92
closed 2023-09-30 by ThomasGerstenberg