ThomasGerstenberg/blatann #93: Pairing fails with "Confirm Value Failed (0x04)"

trembel, 2022-01-19:
I'm trying to connect to a peripheral, which supports passkey pairing, however, I'm not able to pair.

Attached you can find three wireshark recordings (using nRF Sniffer). With the nRF Connect App the pairing works, using Blatann it fails (same dongle with same firmware).

The log files should have the extension `.pcapng` (not allowed by github, thus renamed to`.txt`

The file `blatann_capture.pcapng` differs from the `nrfnonnect_capture.pcapng` during the pairing request in the fields "MITM flag = true" and "Initiator IRK  flag = true". After disabling these (IRK flag manually in `smp.py`), the issue still persists as can be seen in `blatann_capture2.pcapng`.

It seems to me, as if the `Sent Pairing Random` random value is wrong, and thus not confirmed by the peripheral.

Best,
trembel

P.S. In `blatann.log` is the logging output of blatann attached.
[blatann.log](https://github.com/ThomasGerstenberg/blatann/files/7895618/blatann.log)
[blatann_capture.txt](https://github.com/ThomasGerstenberg/blatann/files/7895619/blatann_capture.txt)
[blatann_capture2.txt](https://github.com/ThomasGerstenberg/blatann/files/7895620/blatann_capture2.txt)
[nrfconnect_capture.txt](https://github.com/ThomasGerstenberg/blatann/files/7895621/nrfconnect_capture.txt)

ThomasGerstenberg, 2022-01-21:
Thanks for the logs and captures! I'll take a look at the issue when I can

thaismidorikawa, 2022-03-17:
I'm having a similar issue. Have you managed to pair with your peripheral @trembel?

ThomasGerstenberg, 2022-03-18:
@thaismidorikawa are you also running on Linux? If so, could you try reproducing the issue on Windows? I've seen issues with Linux in the past around pairing, wondering if it's related

trembel, 2022-03-18:
I haven't managed to get it working and, at that time, I also tried it on Windows where it aborted with the same issue.

ThomasGerstenberg, 2022-03-18:
Can you try recapturing and calling `peer.security.use_debug_lesc_key()` prior to pairing?

thaismidorikawa, 2022-03-18:
I noticed that even though I'm registering the on_passkey_required callback it's not being called, so the connection is dropped with a timeout. Which is the same btw for the on_peripheral_security_request

thaismidorikawa, 2022-03-18:
@ThomasGerstenberg With IO Capabilities set to none and the device accepting Just Works pairing method it's being able to connect and bond. But with Passkey it fails.
It does seem to be something when resolving the passkey. It keeps on giving me authentication failure error...
I wish I could check what passkey it's being received but I can't get it from the peripheral unfortunately.

thaismidorikawa, 2022-03-18:
Ok, after trying a couple of things it turns out that the following fixes it:
In blatann/nrf/nrf_driver.py:
```
def ble_gap_auth_key_reply(self, conn_handle, key_type, key):
       key_pointer = None
       if key is not None:
            key_buf = util.list_to_uint8_array(key)
            key_pointer = key_buf.cast()             # fix
        else:
            key_buf = None
        return driver.sd_ble_gap_auth_key_reply(self.rpc_adapter, conn_handle, key_type, key_pointer)
```
Just moving .cast() to be called later seems to fix this. 
I have no clue why this is the case but it solved the issue for me

ThomasGerstenberg, 2022-03-20:
Ok this seems to be the same type of issue I've seen in the past with linux (see 77864de), surprised it wasn't working on Windows also. I'm guessing it might be something to do with how python handles intermediate variables, maybe they're descoped too quickly and need a local to persist through the function? Not sure, best theory I can come up with.

I'll add this change in and have you retest, make sure it all works. I see a couple other spots where this change should probably be made also

closed 2022-05-03 by ThomasGerstenberg