ThomasGerstenberg/blatann #103: Reopening device fails

martinhansdk, 2022-03-25:
I find that I frequently get a timeout when trying to open a BleDevice that I have recently closed. This happens when I'm running a script, interrupt it and then start it again.

I am using a nrf52840 dongle that I've programmed with the connectivity firmware that ships with blatann version 0.3.6. I am on Ubuntu 21.10.

I have boiled the situation down to the following minimal script which simply opens and closes the device in a loop:

```python
#!/usr/bin/env python3
"""Try opening and closing a bunch of times"""

import argparse
from pathlib import Path

from blatann import BleDevice
from blatann.utils import setup_logger

def main():
    "Run as a script."

    parser = argparse.ArgumentParser(description=__doc__, formatter_class=argparse.RawTextHelpFormatter)
    parser.add_argument('-d', '--dongle', required=True, help='BLE dongle serial port')

    args = parser.parse_args()

    setup_logger(level="DEBUG")

    while True:
        print("---------------------")
        assert Path(args.dongle).exists()
        dongle = BleDevice(args.dongle, log_driver_comms=True)
        dongle.open(clear_bonding_data=True)
        dongle.close()

if __name__ == "__main__":
    main()
```

Here is the output that I get from running the script:

```
./debug_timeout.py --dongle=/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01
---------------------
[2022-03-25 13:25:56,898] [MainThread] [blatann.device.clear_bonding_data:207] [INFO]: Clearing out all bonding information
[2022-03-25 13:25:56,899] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] open()
[2022-03-25 13:25:57,303] [MainThread] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [2]: Successfully opened /dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01. Baud rate: 1000000. Flow control: none. Parity: none.
[2022-03-25 13:25:57,304] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]: State change: STATE_START -> STATE_RESET
[2022-03-25 13:25:57,304] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        1 ->  [N/A] type:          RESERVED_5 reliable: no seq#:0 ack#:0 payload_length:0 data_integrity:0 err_code:0x0
[2022-03-25 13:25:57,304] [Dummy-3] [blatann.nrf.nrf_driver._status_handler:654] [INFO]: STATUS [6]: Target Reset performed
[2022-03-25 13:25:57,604] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]: State change: STATE_RESET -> STATE_UNINITIALIZED
[2022-03-25 13:25:57,604] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        2 ->  [01 7e ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0x0 [SYNC]
[2022-03-25 13:25:57,605] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:     1/ 0 <-  [02 7d ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0x0 [SYNC_RESP]
[2022-03-25 13:25:57,605] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]: State change: STATE_UNINITIALIZED -> STATE_INITIALIZED
[2022-03-25 13:25:57,605] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        3 ->  [03 fc 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:25:57,705] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        4 ->  [03 fc 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:25:57,805] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        5 ->  [03 fc 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:25:57,906] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        6 ->  [03 fc 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:25:58,003] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:     2/ 0 <-  [01 7e ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0x0 [SYNC]
[2022-03-25 13:25:58,003] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        7 ->  [02 7d ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0x0 [SYNC_RESP]
[2022-03-25 13:25:58,006] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        8 ->  [03 fc 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:25:58,007] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:     3/ 0 <-  [03 fc 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:25:58,007] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        9 ->  [04 7b 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG_RESP [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:25:58,107] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:       10 ->  [03 fc 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:25:58,107] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:     4/ 0 <-  [04 7b 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG_RESP [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:25:58,108] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]: State change: STATE_INITIALIZED -> STATE_ACTIVE
[2022-03-25 13:25:58,108] [Dummy-3] [blatann.nrf.nrf_driver._status_handler:654] [INFO]: STATUS [7]: Connection active
[2022-03-25 13:25:58,109] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] ble_conn_configure(<blatann.nrf.nrf_types.config.BleConnConfig object at 0x7f2731a22460>,)
[2022-03-25 13:25:58,109] [MainThread] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:       11 ->  [00 69 20 00 00 00 01 01 04 06 00 ] type:     VENDOR_SPECIFIC reliable:yes seq#:0 ack#:0 payload_length:b data_integrity:1 header_checksum:82 err_code:0x0
[2022-03-25 13:25:58,110] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:     5/ 0 <-  [N/A] type:                 ACK reliable: no seq#:0 ack#:1 payload_length:0 data_integrity:0 err_code:0x0
[2022-03-25 13:25:58,110] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:     6/ 0 <-  [01 69 00 00 00 00 ] type:     VENDOR_SPECIFIC reliable:yes seq#:0 ack#:1 payload_length:6 data_integrity:1 header_checksum:ca err_code:0x0
[2022-03-25 13:25:58,110] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:       12 ->  [N/A] type:                 ACK reliable: no seq#:0 ack#:1 payload_length:0 data_integrity:0 err_code:0x0
[... 47 lines cut ...]
[2022-03-25 13:25:58,125] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] close()
[... cut ...]
  File "/home/martin/.local/lib/python3.9/site-packages/blatann/nrf/nrf_driver.py", line 80, in wrapper
    raise NordicSemiException('Failed to {}. {}'.format(wrapped.__name__, err_string), err_code)
pc_ble_driver_py.exceptions.NordicSemiException: Failed to open. Error code: NrfError.rpc_h5_transport_state
```

As you can see, it complains about a non-existing file when trying to open the device. My script checks that the serial port device exists before opening, so I don't know what could cause this.

martinhansdk, 2022-03-25:
I can add, that adding a 10 second sleep after each call to `close()` does help somewhat. Now it succeeds at opening the device a few times before failing with a different error, this time a timeout:

```
---------------------
[2022-03-25 13:39:00,067] [MainThread] [blatann.device.clear_bonding_data:207] [INFO]: Clearing out all bonding information
[2022-03-25 13:39:00,067] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] open()
[2022-03-25 13:39:00,471] [MainThread] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [2]: Successfully opened /dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01. Baud rate: 1000000. Flow control: none. Parity: none.
[2022-03-25 13:39:00,471] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:     1/ 0 <-  [01 7e ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0x0 [SYNC]
[2022-03-25 13:39:00,471] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]: State change: STATE_START -> STATE_RESET
[2022-03-25 13:39:00,472] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        1 ->  [N/A] type:          RESERVED_5 reliable: no seq#:0 ack#:0 payload_length:0 data_integrity:0 err_code:0x0
[2022-03-25 13:39:00,472] [Dummy-4] [blatann.nrf.nrf_driver._status_handler:654] [INFO]: STATUS [6]: Target Reset performed
[2022-03-25 13:39:00,772] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]: State change: STATE_RESET -> STATE_UNINITIALIZED
[2022-03-25 13:39:00,772] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        2 ->  [01 7e ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0x0 [SYNC]
[2022-03-25 13:39:00,773] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:     2/ 0 <-  [02 7d ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0x0 [SYNC_RESP]
[2022-03-25 13:39:00,773] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]: State change: STATE_UNINITIALIZED -> STATE_INITIALIZED
[2022-03-25 13:39:00,773] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        3 ->  [03 fc 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:39:00,874] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        4 ->  [03 fc 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:39:00,974] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        5 ->  [03 fc 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:39:01,074] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        6 ->  [03 fc 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:39:01,172] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:     3/ 0 <-  [01 7e ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0x0 [SYNC]
[2022-03-25 13:39:01,172] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        7 ->  [02 7d ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:2 data_integrity:0 err_code:0x0 [SYNC_RESP]
[2022-03-25 13:39:01,175] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        8 ->  [03 fc 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:39:01,175] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:     4/ 0 <-  [03 fc 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:39:01,176] [Dummy-3] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:        9 ->  [04 7b 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG_RESP [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[2022-03-25 13:39:01,275] [Dummy-4] [blatann.nrf.nrf_driver._log_message_handler:658] [INFO]: LOG [1]:       10 ->  [03 fc 11 ] type: LINK_CONTROL_PACKET reliable: no seq#:0 ack#:0 payload_length:3 data_integrity:0 err_code:0x0 [CONFIG [ sliding-window-size:1 out-of-frame:0 data-integrity-check-type:1 version-number:0 ]]
[... cut ...]
  File "/home/martin/work/marco_recorder/tools/debug_timeout.py", line 26, in main
    dongle.open(clear_bonding_data=True)
  File "/home/martin/.local/lib/python3.9/site-packages/blatann/device.py", line 185, in open
    self.ble_driver.open()
  File "/home/martin/.local/lib/python3.9/site-packages/blatann/nrf/nrf_driver.py", line 80, in wrapper
    raise NordicSemiException('Failed to {}. {}'.format(wrapped.__name__, err_string), err_code)
pc_ble_driver_py.exceptions.NordicSemiException: Failed to open. Error code: NrfError.timeout
```

ThomasGerstenberg, 2022-03-30:
Thank you for the detailed logs! This looks to be similar to #75. It's a known issue that there's some quirks around re-opening the device (especially the USB dongle) that I haven't investigated in detail. I haven't prioritized this issue as the typical/most common use-cases close the device when the program terminates, but I will keep it on my radar.

regin-mBohm, 2023-05-24:
Any updates on this? 
I have the same issue. around 6 seconds delay in between and its very flaky. would be nice with a workaround

ThomasGerstenberg, 2023-06-07:
I haven't looked much into this. I think there's a solution polling pc-ble-driver-py's [enum_serial_ports](https://github.com/NordicSemiconductor/pc-ble-driver-py/blob/a6b36d3a8924b809fe86267ba3f8079dc173fb27/pc_ble_driver_py/ble_driver.py#L2014). I haven't been working in this domain recently so this has fallen to the wayside for a bit

mdxs, 2023-11-08:
The issue can be reduced by waiting for the serial port to be ready before opening the Blatann BLE Device. In the following code, modified from the original test script by @martinhansdk, this is done using [pyserial 3.5](https://pypi.org/project/pyserial/) to find the serial port first, then checking that the serial device path exists:

'''
#!/usr/bin/env python3
"""Try opening and closing a bunch of times"""

import argparse
from pathlib import Path
import serial.tools.list_ports as slp
import time

from blatann import BleDevice
from blatann.utils import setup_logger

logger = setup_logger(level="DEBUG")

def wait_for_serial_port(serial_port, retries=10, delay=0.5):
    logger.info("Wait for serial port {} to be ready...".format(serial_port))
    for retry in range(retries):
        logger.info("run/retry = {}".format(retry))
        for port in slp.comports():
            if port.device == serial_port:
                logger.info("Found with usb_info={}".format(port.usb_info()))
                # HACK: add just a little more time...
                # as sometimes it is found while path device is not yet ready
                time.sleep(delay)
                if Path(serial_port).exists():
                    return True
                logger.info("... but path didn't exist (yet)")
        time.sleep(delay)
    return False

def get_ble_device(serial_port, retries=10, delay=0.5):
    if not wait_for_serial_port(serial_port, retries=retries, delay=delay):
        return None
    ble_device = BleDevice(serial_port, log_driver_comms=True)
    return ble_device

def main():
    "Run as a script."

    parser = argparse.ArgumentParser(
        description=__doc__, formatter_class=argparse.RawTextHelpFormatter
    )
    parser.add_argument(
        "-p",
        "--port",
        required=True,
        help="Serial port for nRF52 with Connectivity firmware",
    )

    args = parser.parse_args()

    while True:
        logger.info("=" * 40)
        ble_device = get_ble_device(args.port)
        if ble_device:
            logger.info("Opening Blatann BLE Device...")
            ble_device.open(clear_bonding_data=True)
            logger.info("Closing Blatann BLE Device...")
            ble_device.close()
        else:
            logger.warning("Failed to obtain Blatann BLE Device in time")

if __name__ == "__main__":
    main()
'''

In my testing on Linux I found that sometimes the serial port is listed/found while the operating system level serial device path isn't ready (yet). The code tries to detect that situation and does another retry/run to find the serial port (and path). It also adds a little more time before checking the path... which feels a bit like a hack, but this does provide much more reliable use of the nRF52840-Dongle (PCA10059).

This approach does introduce delays (as other fixed/static duration sleep workarounds also do), but it tries to be more reliable with as little delay as possible.

Next to testing with the nRF52840-Dongle (PCA10059), I've also tested the script with the nRF52-DK, a Dev Kit for the nRF52832 (PCA10040), and that also worked fine with the script.

mdxs, 2023-11-08:
On 2023-06-07 @ThomasGerstenberg commented:
> I haven't looked much into this. I think there's a solution polling pc-ble-driver-py's [...]

I researched this (after my work on the above PySerial based approach) to see if it could be used.

However, the nRF52840-Dongle (PCA10059) will not be enumerated by `enum_serial_ports` in `pc-ble-driver-py` as reported in https://github.com/NordicSemiconductor/pc-ble-driver-py/issues/64 and confirmed with the following test script (a modification of the `pc_ble_driver_py/examples/serial_port_enum.py` example):

```
def init(conn_ic_id):
    global BLEDriver
    from pc_ble_driver_py import config

    config.__conn_ic_id__ = conn_ic_id
    # noinspection PyUnresolvedReferences
    from pc_ble_driver_py.ble_driver import BLEDriver

    from pc_ble_driver_py import __version__

    print("pc_ble_driver_py v{}".format(__version__))

def main():
    descs = BLEDriver.enum_serial_ports()
    print("enum_serial_ports: {} serial ports found".format(len(descs)))
    for i, d in enumerate(descs):
        print("\nSerial port #{}:".format(i))
        print("|")
        print('|-  Port: "{}"'.format(d.port))
        print('|-  Manufacturer: "{}"'.format(d.manufacturer))
        print('|-  Serial Number: "{}"'.format(d.serial_number))
        print('|-  PnP ID: "{}"'.format(d.pnp_id))
        print('|-  Location ID: "{}"'.format(d.location_id))
        print('|-  Vendor ID: "{}"'.format(d.vendor_id))
        print('|_  Product ID: "{}"'.format(d.product_id))

init("NRF52")
main()
```

Which on Linux showed the following (incorrect, see below) output:

```
pc_ble_driver_py v0.17.0
enum_serial_ports: 1 serial ports found

Serial port #0:
|
|-  Port: "/dev/ttyACM1"
|-  Manufacturer: "SEGGER"
|-  Serial Number: "000682919562"
|-  PnP ID: ""
|-  Location ID: "/sys/devices/pci0000:00/0000:00:11.0/0000:02:03.0/usb2/2-1/2-1:1.0/tty/ttyACM1"
|-  Vendor ID: "1366"
|_  Product ID: "1015"
```

Note that in my case that output is incorrect, as I had the following two nRF52 Connectivity devices present:

- nRF52840-Dongle (PCA10059) - _not seen, though present as /dev/ttyACM0_
- nRF52-DK, a Dev Kit for the nRF52832 (PCA10040) - _seen on /dev/ttyACM1_

Both have proven to work with Blatann just fine (and thus are known-good nRF52 Connectivity devices).

There might be fixes proposed for the underlying `pc-ble-driver` that might fix `enum_serial_ports` in `pc-ble-driver-py` in the future; though that could still take some time. At the moment, this approach will unfortunately not work.