ThomasGerstenberg/blatann #105: ValueError: 127 is not a valid BLEGapAddrTypes

martinhansdk, 2022-04-08:
I am encountering an error while running blatann's unit tests. I am running on Ubuntu 21.10.

`BLATANN_DEV_1` is set to the USB port presented by a nRF-52840-DK flashed with `connectivity_4.1.4_1m_with_s132_5.1.0.hex`

`BLATANN_DEV_2` is set to the USB port presented by a nRF52840 USB Dongle flashed with `connectivity_4.1.4_usb_with_s132_5.1.0_dfu_pkg.zip`

`BLATANN_TEST_QUICK = 1`

Running `make run-tests` at the root of the cloned repository I get the following:

```
[2022-04-08 09:55:04,985] [MainThread] [blatann.device.clear_bonding_data:207] [INFO]: Clearing out all bonding information
[2022-04-08 09:55:04,985] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] open()
[2022-04-08 09:55:06,191] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_conn_configure(<blatann.nrf.nrf_types.config.BleConnConfig object at 0x7f12ab3c11c0>,)
[2022-04-08 09:55:06,197] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_enable(<blatann.nrf.nrf_types.config.BleEnableConfig object at 0x7f12ab3d30d0>,)
[2022-04-08 09:55:06,207] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_device_name_set(b'nRF5x',)
[2022-04-08 09:55:06,209] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_appearance_set(<Appearance.unknown: 0>,)
[2022-04-08 09:55:06,211] [MainThread] [blatann.device.clear_bonding_data:207] [INFO]: Clearing out all bonding information
[2022-04-08 09:55:06,212] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] open()
[2022-04-08 09:55:07,420] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] ble_conn_configure(<blatann.nrf.nrf_types.config.BleConnConfig object at 0x7f12ab3d37f0>,)
[2022-04-08 09:55:07,423] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] ble_enable(<blatann.nrf.nrf_types.config.BleEnableConfig object at 0x7f12ab3dd640>,)
[2022-04-08 09:55:07,428] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] ble_gap_device_name_set(b'nRF5x',)
[2022-04-08 09:55:07,429] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] ble_gap_appearance_set(<Appearance.unknown: 0>,)
test_advertising_no_scan_data (integrated.test_advertising_data.TestAdvertisingData) ... [2022-04-08 09:55:07,430] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_addr_get()
[2022-04-08 09:55:07,431] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_adv_data_set({<Types.flags: 1>: [6], <Types.complete_local_name: 9>: 'Blatann Test'}, {})
[2022-04-08 09:55:07,433] [MainThread] [blatann.gap.advertising._start:210] [INFO]: Starting advertising, params: 'BLEGapAdvParams'(type: <BLEGapAdvType.scanable_undirected: 2>, interval: 25ms, timeout: 10s, ch mask: 000), auto-restart: False
[2022-04-08 09:55:07,433] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_adv_start('BLEGapAdvParams'(type: <BLEGapAdvType.scanable_undirected: 2>, interval: 25ms, timeout: 10s, ch mask: 000), 1)
[2022-04-08 09:55:07,435] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] ble_gap_scan_stop()
[2022-04-08 09:55:07,436] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] ble_gap_scan_start(ScanParameters(window: 100ms, interval: 100ms, timeout: 4s, active: True),)
[2022-04-08 09:55:11,437] [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01_Event] [blatann.device.on_driver_event:34] [DEBUG]: Got NRF Driver event: GapEvtTimeout(conn_handle=65535, src=<BLEGapTimeoutSrc.scan: 1>)
[2022-04-08 09:55:11,448] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_adv_stop()
[2022-04-08 09:55:11,450] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] ble_gap_scan_stop()
ok
test_advertising_scan_data (integrated.test_advertising_data.TestAdvertisingData) ... [2022-04-08 09:55:11,952] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_addr_get()
[2022-04-08 09:55:11,953] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_adv_data_set({<Types.flags: 1>: [6], <Types.complete_local_name: 9>: 'Blatann Test'}, {<Types.service_16bit_uuid_complete: 3>: [205, 171, 240, 222]})
[2022-04-08 09:55:11,955] [MainThread] [blatann.gap.advertising._start:210] [INFO]: Starting advertising, params: 'BLEGapAdvParams'(type: <BLEGapAdvType.scanable_undirected: 2>, interval: 25ms, timeout: 10s, ch mask: 000), auto-restart: False
[2022-04-08 09:55:11,955] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_adv_start('BLEGapAdvParams'(type: <BLEGapAdvType.scanable_undirected: 2>, interval: 25ms, timeout: 10s, ch mask: 000), 1)
[2022-04-08 09:55:11,957] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] ble_gap_scan_stop()
[2022-04-08 09:55:11,957] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] ble_gap_scan_start(ScanParameters(window: 100ms, interval: 100ms, timeout: 4s, active: True),)
[2022-04-08 09:55:15,959] [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01_Event] [blatann.device.on_driver_event:34] [DEBUG]: Got NRF Driver event: GapEvtTimeout(conn_handle=65535, src=<BLEGapTimeoutSrc.scan: 1>)
[2022-04-08 09:55:15,967] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_adv_stop()
[2022-04-08 09:55:15,969] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] ble_gap_scan_stop()
ok
test_dynamic_adv_data_update (integrated.test_advertising_data.TestAdvertisingData) ... [2022-04-08 09:55:16,471] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_addr_get()
[2022-04-08 09:55:16,472] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_adv_data_set({<Types.service_data: 22>: [171, 205, 0]}, {})
[2022-04-08 09:55:16,474] [MainThread] [blatann.gap.advertising._start:210] [INFO]: Starting advertising, params: 'BLEGapAdvParams'(type: <BLEGapAdvType.non_connectable_undirected: 3>, interval: 25ms, timeout: 20s, ch mask: 000), auto-restart: False
[2022-04-08 09:55:16,474] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_adv_start('BLEGapAdvParams'(type: <BLEGapAdvType.non_connectable_undirected: 3>, interval: 25ms, timeout: 20s, ch mask: 000), 1)
[2022-04-08 09:55:16,476] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] ble_gap_scan_stop()
[2022-04-08 09:55:16,477] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01] ble_gap_scan_start(ScanParameters(window: 100ms, interval: 100ms, timeout: 10s, active: True),)
Exception in thread /dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_D66630A5B594-if01_Event:
Traceback (most recent call last):
[... cut ...]
[2022-04-08 09:55:18,986] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_adv_data_set({<Types.service_data: 22>: [171, 205, 5]}, {})
[2022-04-08 09:55:19,488] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-SEGGER_J-Link_000683632362-if00] ble_gap_adv_data_set({<Types.service_data: 22>: [171, 205, 6]}, {})
...
```

Is there perhaps a device in my vicinity that is reporting something that blatann can not cope with?

ThomasGerstenberg, 2022-04-22:
Looks like there's an `anonymous` BLE address type which is defined in SD v6 of [ble_gap.h](https://github.com/NordicSemiconductor/pc-ble-driver/blob/master/include/sd_api_v6/ble_gap.h#L209) but not in v5.

I'll add support for this so it's recognized, may need to make some other changes to how the GAP address is decoded

martinhansdk, 2022-04-23:
Awesome! Thanks for looking at this. I'll try to identify the guilty device here so I can test your fix when it is ready.

ThomasGerstenberg, 2022-04-28:
Alright I was able to make a basic android app to perform anonymous advertising. However I can't seem to get blatann to pick up the anonymous packet. SoftDevice v5.1 doesn't look to support extended (non-legacy) advertisements, which anonymous packets would be on.

I'm really confused how you're seeing this issue, but regardless I'll put in support so it's decoded fine

ThomasGerstenberg, 2022-04-28:
Hopefully the 127 value isn't a complete coincidence and there isn't some deeper issue going on...

martinhansdk, 2022-04-28:
Awesome!  Thank you.

Yesterday I made a script that printed out the advertising information of the devices that have anonymous addresses and walked around the building while keeping an eye on the rssi. I wasn't able to find the devices, but I could see that there were two seemingly identical devices there plus one device which had the string "signify.com" in the manufacturing data. This appears to be a Philips lighting product of some kind. The other two devices disappeared around the time when most people left work, so it is probably someone's cell phone or headset or something.

There is an Android app called "nRF Connect" on which I could also see the devices. It has a cloning feature so maybe I could use it to collect enough data for you to reproduce it at your end.

I'll also test your fix and see how it fares here.

martinhansdk, 2022-05-02:
I have been trying to spot the offending device again since Thursday, but without luck. Probably the colleague who owns it hasn't been at the office. I'll keep looking.

ThomasGerstenberg, 2022-05-03:
I'll go ahead and merge the associated PR as-is then since it's fairly harmless and close out this issue. If you end up reproducing the issue with the fix I'll reopen this ticket

closed 2022-05-03 by ThomasGerstenberg