ThomasGerstenberg/blatann #107: Installation fails with python 3.10

rds-tmeyer, 2022-04-20:
Hi,

The installation through pip fails with python 3.10.4 with the following message : 
```
(venv) ➜ test_blatan_python_3.10  python --version
Python 3.10.4
(venv) ➜ test_blatan_python_3.10  pip --version
pip 22.0.4 from .\venv\lib\site-packages\pip (python 3.10)
(venv) ➜ test_blatan_python_3.10  pip install blatann
Collecting blatann
  Using cached blatann-0.3.6-py3-none-any.whl (186 kB)
  Downloading blatann-0.3.5-py3-none-any.whl (184 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 184.9/184.9 KB 10.9 MB/s eta 0:00:00
  Downloading blatann-0.3.4-py3-none-any.whl (183 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 183.6/183.6 KB ? eta 0:00:00
  Downloading blatann-0.3.3-py3-none-any.whl (153 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 153.1/153.1 KB 9.5 MB/s eta 0:00:00
  Downloading blatann-0.3.2-py3-none-any.whl (148 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 148.5/148.5 KB 9.2 MB/s eta 0:00:00
  Downloading blatann-0.3.1-py3-none-any.whl (148 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 148.1/148.1 KB ? eta 0:00:00
  Downloading blatann-0.3.0-py3-none-any.whl (144 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 144.5/144.5 KB 8.4 MB/s eta 0:00:00
ERROR: Cannot install blatann==0.3.0, blatann==0.3.1, blatann==0.3.2, blatann==0.3.3, blatann==0.3.4, blatann==0.3.5 and blatann==0.3.6 because these package versions have conflicting dependencies.

The conflict is caused by:
    blatann 0.3.6 depends on pc-ble-driver-py>=0.13
    blatann 0.3.5 depends on pc-ble-driver-py>=0.13
    blatann 0.3.4 depends on pc-ble-driver-py>=0.13
    blatann 0.3.3 depends on pc-ble-driver-py>=0.13
    blatann 0.3.2 depends on pc-ble-driver-py>=0.13
    blatann 0.3.1 depends on pc-ble-driver-py>=0.13
    blatann 0.3.0 depends on pc-ble-driver-py>=0.13

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
```

rds-tmeyer, 2022-04-20:
Relates to https://devzone.nordicsemi.com/f/nordic-q-a/82618/pc-ble-driver-py-release-for-python-3-10

ThomasGerstenberg, 2022-04-22:
As you've pointed out, unfortunately there's nothing I can really do on my end until pc-ble-driver-py supports 3.10. Building the libs myself and removing pc-ble-driver-py as a dep has been on my todo list, but that won't happen for awhile.

I'll leave this issue open until pc-ble-driver-py is updated to support py3.10

ThomasGerstenberg, 2022-07-12:
pc-ble-driver-py has been updated to support Python 3.10

closed 2022-07-12 by ThomasGerstenberg