ThomasGerstenberg/blatann #116: Peer Address not saved in bonding database

rene-becker-setec, 2022-06-06:
Hi,

I am using this library (v0.4.0) to drive a BLE peripheral with security enabled.

With the previous version I could not get pairing to work at all (running on Linux (Kubuntu 22.04), but with v0.4.0 pairing works OK.

However, I am finding that the bond information appears not to be saved properly.

After closing the device I find a file "bonding_db.pkl", and when running my script again I see that blatann is finding & loading this file, but every time I connect to my peripheral the passkey prompt is executed again.

When inspecting the saved bond database file I see that the peer address field is not correct. It's "00:00:00:00:00:00".

When I edit the bond database file (with an auxiliary script which load/un-pickles the data, fixes the peer_address and then pickles/saves the data again) everything works as expected, i.e. I am not seeing the passkey entry routine called anymore.

ThomasGerstenberg, 2022-06-14:
Thanks for the report, I'll take a look

ThomasGerstenberg, 2023-02-26:
May be about 8 months too late, but I've updated the bond database format from pickle to json in #133. Hopefully this resolves the issue you were seeing. I wasn't able to reproduce the issue myself on Windows, didn't try linux though

rene-becker-setec, 2023-02-26:
Awesome, thanks. Will hopefully be able to give it a spin soon

ThomasGerstenberg, 2023-09-29:
Closing this issue as no longer reproducible with the JSON format. Can reopen the issue in the future if it's reproduced again

closed 2023-09-29 by ThomasGerstenberg