ThomasGerstenberg/blatann #117: Handling multiple nRF Dongles and/or DKs

rene-becker-setec, 2022-06-06:
Hi,

Maybe a rather unusual use-case, but many of our products acting as BLE peripherals support multiple concurrent connections by independent BLE centrals. But then again, Nordic/Zephyr support this for their devices, so there must be sufficient interest in such configuration... 

I was hoping to make use of this library for developing a test-environment emulating multiple centrals by just using multiple dongles plugged into a single PC.

Being able to create multiple BleDevice() instances and then controlling them from threads would be great, but even if this lib is not thread safe driving the instance sequentially from 1 thread will already go a long way.

Connecting with multiple (2) dongles does appear to work, but I am having trouble subscribing to notifications. If I subscribe to notifications blatann appears to only register notifications received for the most recent connected dongle.

I am not sure whether the underlying pc-ble-driver-py library supports handling multiple devices, but driving multiple dongles via Nordics GUI application appears to work, so I am hopeful.

ThomasGerstenberg, 2022-06-14:
Can you provide output with debug logging enabled? It'll help track down what's going on. The library should be able to support multiple dongles no problem

Easiest way to enable logging:
```python
from blatann.utils import setup_logger
setup_logger("blatann", "DEBUG")
```

rene-becker-setec, 2022-06-14:
Sure, will do. Will only be able to do this beginning of next week though.

rene-becker-setec, 2022-06-22:
I have tried to capture logs - unfortunately (?!) this time it appears to work ok ... i.e. I am seeing both dongles reporting notifications. I have not changed the scripts, but had to re-program the dongles.

I am not sure whether the FW programmed on the dongles is identical - I just use the nRF Connect Desktop app to program my dongles and as far as I remember I have seen an update. Not sure the the FW was changed though and have no record/logs that detail the FW version.

Anyway, consider it a false alarm for now ... closing this ticket.

Thanks for your support

rene-becker-setec, 2022-06-22:
I might have been a bit too quick - after running a few more times I am seeing this again.

Attached is a log for when all was working as expected as well as when I only receive notifications on one dongle.

In the problematic log I see that dongle 'CDD43EAD0F05' (successfully) subscribes to the characteristic of interest, but no notifications thereafter. I am quite confident that the target device (GATT server) does emit notifications, but at the moment have not got enough dongles/dev kits to also run a connection sniffer at the same time to confirm. I will try to do this later this week, just to be sure.

```
[2022-06-22 17:50:55,930] [MainThread] [blatann.nrf.nrf_driver.wrapper:64] [DEBUG]: [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_CDD43EAD0F05-if01] ble_gattc_write(0, BLEGattcWriteParams(handle=37, write_op=<BLEGattWriteOperation.write_req: 1>, offset=0, flags=<BLEGattExecWriteFlag.prepared_cancel: 0>, data=b'\x01\x00'))
[2022-06-22 17:50:55,987] [/dev/serial/by-id/usb-Nordic_Semiconductor_nRF52_Connectivity_CDD43EAD0F05-if01_Event] [blatann.device.on_driver_event:34] [DEBUG]: Got NRF Driver event: GattcEvtWriteResponse(conn_handle=0, status=<BLEGattStatusCode.success: 0>, error_handle=0, attr_handle=37, write_of=<BLEGattWriteOperation.write_req: 1>, offset=0, data='')

```

[2022-06-22_notifications_from_one_dongle_only.log](https://github.com/ThomasGerstenberg/blatann/files/8955647/2022-06-22_notifications_from_one_dongle_only.log)
[2022-06-22_1_good.log](https://github.com/ThomasGerstenberg/blatann/files/8955648/2022-06-22_1_good.log)

ThomasGerstenberg, 2022-06-27:
In the logs with the issue, I'm not seeing notifications being received from pc-ble-driver at all for the first connection created with the device. It would be good to see sniffer logs to ensure the device is sending notifications

rene-becker-setec, 2022-06-28:
Yes, it certainly looks that way.
I am reasonably confident that the device is emitting notifications if configured, but indeed an on-air activity trace would provide more insights. I will try to do this, but will need to gather additional HW first.

I also tried driving the 2 dongles with 2 instances of the nRF Connect Desktop App. Whilst this always worked, i.e. I was able to observe notifications, I did notice some odd behavior - the 2nd instance started reporting notifications right away without me having to subscribe.

ThomasGerstenberg, 2023-01-26:
@rene-becker-setec any update on this? Otherwise I'll close as a non-issue

rene-becker-setec, 2023-01-27:
Hi @ThomasGerstenberg,

It still is an issue, but unfortunately I have not had a chance to spend much time on it.
Feel free to close - if I get around to it and find more information I can re-open it again ;-)

Thanks.

closed 2023-01-27 by ThomasGerstenberg