ThomasGerstenberg/blatann #119: scanner.py doesn't work on Windows 10

snowuyl, 2022-07-07:
nRF Connect can scan BLE device. But the following doesn't have any response.
D:\Python\Python37\python.exe .\scanner.py

snowuyl, 2022-07-07:
[image]
[scanner.zip](https://github.com/ThomasGerstenberg/blatann/files/9060686/scanner.zip)

snowuyl, 2022-07-07:
[image]

ThomasGerstenberg, 2023-01-26:
The command line needs a second parameter with the name of the port, e.g. `D:\Python\Python37\python.exe .\scanner.py COM5`. This has been tested on windows extensively and should work as expected

closed 2023-01-26 by ThomasGerstenberg