ThomasGerstenberg/blatann #142: Is it possible to suppress saving bond database?

mdxs, 2023-09-26:
I'm looking into using Blatann with an nRF52840 dongle to interact with BLE devices in a testing environment in which the bond database isn't expected to add any value. Is it possible to suppress saving the bond database?

Currently, as I understand it and see from the logs, there is always an attempt to save the bond database. It probably also attempts to load the bond database as well. In my case the saved file is a JSON dict with an empty "records" list, and I expect this to remain empty in my usage scenario of Blatann.

I'm thinking that a special DatabaseStrategy might work (one that simply doesn't try to do any file stuff), but I guess there should also be some changes to the initialization phase to specify the use of such a "NoneDatabaseStrategy/MemoryDatabaseStrategy".

mdxs, 2023-09-26:
Perhaps using the special filename ":memory:" to create an in-memory database (similar to SQLite).

ThomasGerstenberg, 2023-09-26:
I like it, should be pretty straight-forward to add in a new strategy for that. The `:memory:` designator is a good suggestion.

I may have time in the next week or so to add this in. In the meantime you can work around this by overriding the current DB loader in the device to one that does nothing on load/save

```py
from blatann.gap.bond_db import BondDatabaseLoader
from blatann.gap.default_bond_db import DefaultBondDatabase

class NoopBondDbLoader(BondDatabaseLoader):
    def load(self):
        return DefaultBondDatabase()
    def save(self, db):
        # do nothing
        pass

device = BleDevice()
device.bond_db_loader = NoopBondDbLoader()
device.configure()
device.open()
```

mdxs, 2023-09-26:
Thanks for the workaround, it works for my use case.

Looking forward to some `:memory:` database implementation.

closed 2023-09-27 by ThomasGerstenberg

mdxs, 2023-10-26:
Great to see that this has been implemented.

PR #144 (2023-09-27): #142: adds in-memory bond database strategy
Adds special bond db filename `:memory:` to signify the bond database should not be written to disk

Fixes #142