ThomasGerstenberg/blatann #150: Conflicting Dependencies on Install

HVIEH, 2023-10-03:
Apologies if this is a trivial issue, but when I try to install the package with "pip install blatann" I receive this:
    ERROR: Cannot install blatann==0.3.0, blatann==0.3.1, blatann==0.3.2, blatann==0.3.3, blatann==0.3.4, blatann==0.3.5, blatann==0.3.6, blatann==0.4.0 and blatann==0.5.0 because these package versions have conflicting dependencies.
    
    The conflict is caused by:
        blatann 0.5.0 depends on pc-ble-driver-py>=0.13
        blatann 0.4.0 depends on pc-ble-driver-py>=0.13
        blatann 0.3.6 depends on pc-ble-driver-py>=0.13
        blatann 0.3.5 depends on pc-ble-driver-py>=0.13
        blatann 0.3.4 depends on pc-ble-driver-py>=0.13
        blatann 0.3.3 depends on pc-ble-driver-py>=0.13
        blatann 0.3.2 depends on pc-ble-driver-py>=0.13
        blatann 0.3.1 depends on pc-ble-driver-py>=0.13
        blatann 0.3.0 depends on pc-ble-driver-py>=0.13

Are they not all depending on the same versions here? Do I need to specify the version of blatann? When I try "pip install blatann "package == 0.5.0" I get:
    ERROR: Cannot install blatann==0.3.0, blatann==0.3.1, blatann==0.3.2, blatann==0.3.3, blatann==0.3.4, blatann==0.3.5, blatann==0.3.6, blatann==0.4.0 and blatann==0.5.0 because these package versions have conflicting dependencies.
    
    The conflict is caused by:
        blatann 0.5.0 depends on pc-ble-driver-py>=0.13
        blatann 0.4.0 depends on pc-ble-driver-py>=0.13
        blatann 0.3.6 depends on pc-ble-driver-py>=0.13
        blatann 0.3.5 depends on pc-ble-driver-py>=0.13
        blatann 0.3.4 depends on pc-ble-driver-py>=0.13
        blatann 0.3.3 depends on pc-ble-driver-py>=0.13
        blatann 0.3.2 depends on pc-ble-driver-py>=0.13
        blatann 0.3.1 depends on pc-ble-driver-py>=0.13
        blatann 0.3.0 depends on pc-ble-driver-py>=0.13

Appreicate any input, thank you.

HVIEH, 2023-10-03:
Ah I thought I had the most up to date version of that package but I only have 11.4. For some reason I can only install versions up to 11.4 in PyCharm even though it lists them.

ThomasGerstenberg, 2023-10-04:
PyCharm may think that you're using Python2.7, iirc 11.4 is the latest supported for Py2.7. Double-check your pycharm settings to make sure it's using the right interpreter

HVIEH, 2023-10-04:
It has to be somehow related to the python version I think, but I only have python3 on my computer and pycharm is using it. I tried "python3 -m install pc-ble-driver-py>=0.13" and got this:

[image]

Not really sure what to do about this.

HVIEH, 2023-10-10:
I solved this by installing python 3.10.11. I had been trying to use a 3.12 version and later tried a 3.11 version of python when I was getting this problem.

closed 2023-10-10 by HVIEH