ThomasGerstenberg/blatann #191: Providing tk when using legacy pairing and oob

shiftreg, 2025-06-26:
As the title says. I'm currently try to use legacy pairing via OOB but i cannot find a documented API where to set the required OOB data.
Can anyone support/hint with this topic?

ThomasGerstenberg, 2025-07-04:
OOB pairing isn't supported currently. I'll take a look at what the effort would be to add it in

shiftreg, 2025-07-15:
@ThomasGerstenberg i found out that i could realise OOB pairing with using the following event.

`    def _on_passkey_entry(self, peer, event_args: PasskeyEntryEventArgs):
        """
        Event callback for when the client disconnects from us (or when we disconnect from the client)

        :param peer: The peer that disconnected
        :type peer: blatann.peer.Client
        :param event_args: The event args
        :type event_args: blatann.event_args.DisconnectionEventArgs
        """
        logging.info("on_passkey_entry: {}".format(peer.peer_address))
        peer.security.ble_device.ble_driver.ble_gap_auth_key_reply(peer.conn_handle, event_args.key_type, bytes.fromhex(self.oob3_tk))
`

Pairing works fine now. But now iam encountering a new issue. The bonding information will not be stored within the database due to an already existing entry. Does the id need to be incremented at some point?

[image]